arduino/uno-r4-wifi-usb-bridge #9: 0.2.0 Firmware update has bricked my Arduino R4 Wifi

PrinceBalabis, 2023-07-13:
Bug description:
After flashing successfully to 0.2.0. I am no longer able to upload any code to the Arduino R4 Wifi. I am getting error message "No device found on cu.usbmodemDC5475C5165C2". I have tried to flash and upload code from my Macbook and my Windows PC but I am having the same issue on both computers.
See below image to see error message I am receiveing when upload code from Arduino IDE.
[image]

Reproduce:
1. Firmware updated using instructions from [this blog](https://forum.arduino.cc/t/radio-module-firmware-version-0-2-0-is-now-available/1147361). The firmware update is successfully completed.
3. Try to Upload code from Arduino IDE but fails with error message "No device found on cu.usbmodemDC5475C5165C2".

facchinm, 2023-07-13:
Hi @PrinceBalabis ,
Did you disconnect physically the board from the computer after updating the firmware? That's unfortunate but there's a bug in esp32 preventing the board to reboot without user intervention.

vshymanskyy, 2023-07-14:
Yes, looks like this happens from time to time, not directly after the firmware upgrade.
Reconnecting the board physically helps.

PrinceBalabis, 2023-07-14:
> 

I have physically reconnected the board to the Macbook after flashing firmware, even tried to connect to my PC afterwards, getting same error message on both computers. To clarify, I can find the board on the list of devices in Arduino IDE on both my Macbook and PC, but it's only when uploading code that I get the error message.

LYB030, 2023-07-16:
i got same problem

PrinceBalabis, 2023-07-21:
I bought a 2nd Arduino Uno R4 and tried to flash using the guide on [Arduino Cloud IoT](https://create.arduino.cc/iot/). This broke my 2nd Arduino Uno R4 as well so I have two Arduino Uno R4 which are no longer working.

RickorDD, 2023-07-21:
i flashed with the Jumper on ESP Pin and flashed with MacOS and ./bin/espflash ... and it was successful
With Windows has not running with error and the workaround with Mac Console was not running without jumper.
After the flash i press the reset button, not USB unplugged

But now I also have the problem that the UNO R4 sometimes does not start immediately and hangs. But after reset button it works

pennam, 2023-07-21:
@PrinceBalabis @RickorDD are your boards from the early access program?

@PrinceBalabis have you tried using a different PC and/or USB cable? What macOS version are you using?

RickorDD, 2023-07-21:
no board from early access

pennam, 2023-07-21:
> After the flash i press the reset button, not USB unplugged

After flashing the wifi firmware you have to unplug the board, remove the jumper wire, and connect the board again to the PC. Only pressing the reset button the board will stay in ESP download mode.

PrinceBalabis, 2023-07-21:
> @PrinceBalabis @RickorDD are your boards from the early access program?
No

> @PrinceBalabis have you tried using a different PC and/or USB cable? What macOS version are you using?
Yes, I have tried flashing firmware and uploading code from all of these devices:
1.  Macbook #1
2.  Macbook #2
3.  Windows PC

I follow these instructions (including physically disconnecting the USB cable and trying to upload code from all my Macbooks and PC's)
[image]

See here that the flash is successful from my PC:
[image]

See here how the upload fails (same with my PC, Mac #1 and Mac #2):
[image]

LYB030, 2023-07-21:
same problem

RickorDD, 2023-07-21:
do you unplugged the jumper? With Jumper is COM5 without COM4

RickorDD, 2023-07-22:
Update to hang.
UNO R4 starts ever, but sometimes is to long that connect to WiFi
I will check this again.

PrinceBalabis, 2023-07-22:
> do you unplugged the jumper? With Jumper is COM5 without COM4

I don't use a jumper when flashing 0.2.0. It flashes fine. When I replug the Arduino it is COM4, and when I try to flash it fails. see my previous comment with pictures showing that it is COM5 when flashing and COM4 when uploading code: https://github.com/arduino/uno-r4-wifi-usb-bridge/issues/9#issuecomment-1646102559

LYB030, 2023-07-22:
[image]
same problem!

RickorDD, 2023-07-22:
when could be use a jumper or when not??? Whats say the Arduino Team? Can you check with jumper?

[image]

RickorDD, 2023-07-22:
[image]

pennam, 2023-07-24:
@PrinceBalabis i've prepaerd a package of the `0.1.0` firmware version you can upload using the update script:

* https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/download/0.1.0/unor4wifi-update-macos-0.1.0.zip
* https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/download/0.1.0/unor4wifi-update-linux-0.1.0.zip
* https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/download/0.1.0/unor4wifi-update-windows-0.1.0.zip

can you please check if going back to firmware version 0.1.0 restores upload functionality of your boards?

pennam, 2023-07-24:
@RickorDD The jumper wire is used to put the board in ESP downaload mode. The same job can be done using the `unor4wifi-reboot-*` binary that is included in the `unor4wifi-update-*` package.

You should not use the jumper wire if you are using the `unor4wifi-update-*` package.

If there are troubles using the `unor4wifi-update-*` package (board is not going in ESP download mode) You can manually put the board in in ESP download mode using the jumper wire following this istructions:

https://github.com/arduino/uno-r4-wifi-usb-bridge/tree/main/unor4wifi-updater#option-2

PrinceBalabis, 2023-07-24:
> @PrinceBalabis i've prepaerd a package of the `0.1.0` firmware version you can upload using the update script:
>
> * [...]

Thank you for the help! Can confirm I flashed both my Arduinos R4's and I can now upload code just fine.

1. Downloaded the macos 0.1.0 zip file you linked
2. Ran chmod +x update.command
3. Ran ./update.command. 
4. The flash was successful
5. Unplug and replug Arduino R4
6. Upload code from Arduino IDE
7. Upload successful!

[image]

[image]

PrinceBalabis, 2023-07-25:
Now this is interesting. After flashing to 0.1.0, I thought I'll try to flash 0.2.0 again. I did the same procedure and now uploading code to my 0.2.0 Arduino works fine. I tried the same with flashing my 2nd Arduino, and it works fine too.

I can only think of two theories why my Arduinos bricked:
1. I had almost the exact same code on both my Arduino R4's when I flashed them (both Arduinos completely unattached from any components other than the USB cable so it wasn't hardware related). Perhaps the uploaded code somehow affected the flash and caused the brick. However, this doesn't exactly explain why the Arduinos started working again after flashing 0.1.0, after trying to reflash 0.2.0 multiple times and they remained bricked. So this theory is maybe less likely.
2. Flashing 0.2.0 on a vanilla production Arduino R4 bricks it. Flashing with the 0.1.0 firmware first somehow allows for 0.2.0 to flash successfully.

pennam, 2023-07-25:
@PrinceBalabis Thanks for the feedback `0.1.0` is the vanilla production firmware so i suspect the issue could be in the sketch loaded during the upgrade from `0.1.0` to `0.2.0`. Do you remember what sketch was running when your board has bricked?

PrinceBalabis, 2023-07-26:
> @PrinceBalabis Thanks for the feedback `0.1.0` is the vanilla production firmware so i suspect the issue could be in the sketch loaded during the upgrade from `0.1.0` to `0.2.0`. Do you remember what sketch was running when your board has bricked?

I'm not completely sure of the exact code because I don't have a commits on the code before flashing, just approximately.
I had my own code uploaded used for a quadcopter project, though because it was very early stage of development, it was not a lot of code involved when I was flashing the first Arduino, and looking back at the commit history, I believe it was quite a different code from when I flashed the 2nd Arduino.

I believe when I flashed the first Arduino, it was quite lightweight, it involved Serial, FreeRTOS & blinking built-in LED, I might've had some WiFi code then, not sure. 

The code I had uploaded when I flashed the 2nd Arduino, No longer had FreeRTOS but included Wire.h and some code for communicating with MPU-9150 and MP-280.

princelundgren, 2023-08-13:
Flashed with latest 0.2.1 firmware on both my Arduino Uno R4 and uploading sketch works fine! https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/tag/0.2.1

closed 2023-08-13 by princelundgren