arduino/uno-r4-wifi-usb-bridge #11: Wifi key longer than 32 digits not supported

michi2005, 2023-07-29:
Dear all

I tried to connect my Arduino UNO R4 WIFI to the Arduino Cloud,
but as soon as the Wifi passphrase is longer than 32 the following error occurs:

```
***** Arduino IoT Cloud - configuration info *****
Device ID: 3a7e527b-c2e5-4e18-b5f1-c3aa1bf41e92
MQTT Broker: mqtts-up.iot.arduino.cc:8884
WiFi.status(): 0
Current WiFi Firmware: 0.2.1

Firmware name: "/tmp/arduino-build-D1DE31BE964924FF0270A6CC62C6FB08/sketch_jul19a.ino", compiled on: Jul 26 2023
Fault on interrupt or bare metal(no OS) environment
===== Thread stack information =====
  addr: 20007e20    data: 20000c3c
  addr: 20007e24    data: 20000c48
  addr: 20007e28    data: 20007e58
  addr: 20007e2c    data: 00007c5b
  addr: 20007e30    data: 20000c3c
  addr: 20007e34    data: 20007e58
  addr: 20007e38    data: 00017c01
  addr: 20007e3c    data: 00016f42
  addr: 20007e40    data: 00016f02
  addr: 20007e44    data: 20000c94
  addr: 20007e48    data: 20003064
  addr: 20007e4c    data: 05000000
  addr: 20007e50    data: 00008171
  addr: 20007e54    data: 00017c0e
  addr: 20007e58    data: 00017c01
  addr: 20007e5c    data: 00016f42
  addr: 20007e60    data: 00016f02
  addr: 20007e64    data: 00000000
  addr: 20007e68    data: 20002110
  addr: 20007e6c    data: 00000000
  addr: 20007e70    data: 0000001e
  addr: 20007e74    data: 444f4d49
  addr: 20007e78    data: 0d203a45
  addr: 20007e7c    data: 004b4f0a
  addr: 20007e80    data: 20007e88
  addr: 20007e84    data: 0000000a
  addr: 20007e88    data: 4745422b
  addr: 20007e8c    data: 54534e49
  addr: 20007e90    data: 20003a41
  addr: 20007e94    data: 0000000b
  addr: 20007e98    data: 5445472b
  addr: 20007e9c    data: 200002ec
  addr: 20007ea0    data: 00000001
  addr: 20007ea4    data: 0000a500
  addr: 20007ea8    data: ffffffff
  addr: 20007eac    data: 00007465
  addr: 20007eb0    data: 200002ec
  addr: 20007eb4    data: 00007305
  addr: 20007eb8    data: ffffffff
  addr: 20007ebc    data: 2000090c
  addr: 20007ec0    data: 00004040
  addr: 20007ec4    data: 000048e7
  addr: 20007ec8    data: 2000090c
  addr: 20007ecc    data: 00005223
  addr: 20007ed0    data: 00000000
  addr: 20007ed4    data: 0000e191
  addr: 20007ed8    data: 000186e4
  addr: 20007edc    data: 40046f00
  addr: 20007ee0    data: 00000000
  addr: 20007ee4    data: 0000e1cf
  addr: 20007ee8    data: 000186e4
  addr: 20007eec    data: 0000b0fb
  addr: 20007ef0    data: 000186e4
  addr: 20007ef4    data: 0000bb37
  addr: 20007ef8    data: 0000bb2d
  addr: 20007efc    data: 00002599
====================================
=================== Registers information ====================
  R0 : 53646534  R1 : 00017b84  R2 : 00000000  R3 : 20000c9c
  R12: 20007d7b  LR : 00007c5b  PC : 0000dc46  PSR: 41000000
==============================================================
Bus fault is caused by precise data access violation
The bus fault occurred address is 53646534
Show more call stack info by run: addr2line -e "/tmp/arduino-build-D1DE31BE964924FF0270A6CC62C6FB08/sketch_jul19a.ino".elf -a -f 0000dc46 00007c5a 00008170 00007464 00007304 000048e6 00005222 0000e190 0000e1ce 0000b0fa 0000bb36 0000bb2c

```

Can you please tell me how to solve it?

Thanks in advance

cdr-chakotay, 2024-02-23:
Hi, I'm having the exact same issue!

pennam, 2024-03-01:
@michi2005 @cdr-chakotay  could you please re test using the latest ArduinoCore-renesas and wifi firmware?

* https://github.com/arduino/ArduinoCore-renesas/releases/tag/1.1.0
* https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/tag/0.4.1

per1234, 2024-03-01:
@cdr-chakotay later provided an update on their Arduino Forum topic:

https://forum.arduino.cc/t/no-wifi-connection-possible-with-complex-passphrase/1151324/19

>> Which version of the UNOR4 / Renesas boards library do you use?
>>
>>The TXbuff in version 1.0.2 used to be 64 bytes, starting 1.0.4 it became 128.
>
>Exactly this was the issue. Updated to 1.0.4. This worked out !

pennam, 2024-03-01:
Thanks @per1234 so we can close it :partying_face:

closed 2024-03-01 by pennam

cdr-chakotay, 2024-03-03:
Yes, 100 % Solved :)

Thank you for updating the status here @per1234. Initially I planned to update the post here, but I got distracted from that.