arduino/uno-r4-wifi-usb-bridge #16: Compile Error and Arduinoble Version Incompatibility with Arduino Uno R4

TheHafez, 2023-09-04:
I'm encountering issues when trying to compile using compile.sh. Here is the error message I receive:
 _( i used a new arduinoble version -read below- )_

> "C:\\Users\\Hafez\\AppData\\Local\\Arduino15\\packages\\esp32-patched\\esp32/tools/gen_esp32part.exe" -q "C:\\Users\\Hafez\\AppData\\Local\\Temp\\arduino\\sketches\\5B007CB95EE8DB2B76A7289729433FD9/partitions.csv" "C:\\Users\\Hafez\\AppData\\Local\\Temp\\arduino\\sketches\\5B007CB95EE8DB2B76A7289729433FD9/UNOR4USBBridge.ino.partitions.bin"
> Error at line 8: Value 'undefined' is not valid. Known keywords: esphttpd, phy, efuse, nvs, fat, spiffs, ota, coredump, nvs_keys

Also why you are using an [old Arduinoble version in the compile.sh](https://github.com/arduino/uno-r4-wifi-usb-bridge/blob/0.3.0/compile.sh#L15C5-L15C5) ? it does not support uno r4, when I stick to this older version, I get an error stating [HCIVirtualTransport.h](https://github.com/arduino-libraries/ArduinoBLE/blob/56dbfb33aef47489cfbfb26a4e32c8e6d123c9ff/src/utility/HCIVirtualTransport.h) not found in arduinoble/src/utility , thats why I tried upgrading to a newer version of Arduinoble, but that didn't resolve the issue and introduced a the new error I mentioned earlier 

Could you please assist with these issues?

pennam, 2023-09-04:
Hi @TheHafez thanks for the heads up. The branch i was using to build the firmware was force-pushed before the release so it is not working anymore. I've updated the compile.sh script to use the latest ArduinoBLE library. See https://github.com/arduino/uno-r4-wifi-usb-bridge/pull/17

I'm not able to replicate the error with the partition table, but i suspect it is related to this patch https://github.com/arduino/uno-r4-wifi-usb-bridge/blob/bf900770c8af1759b5629d953c851a2756e269fa/core_esp32.patch#L287C1-L316C4

can you please try to delete the `hardware` directory inside the `uno-r4-wifi-usb-bridge` to start from a clean condition and retry?

closed 2023-09-04 by facchinm

TheHafez, 2023-09-04:
Hi @pennam,
I've attempted to run the updated compile.sh script on a clean setup but am still encountering the same error:
```
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
"C:\\Users\\Hafez\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.9/tools/gen_esp32part.exe" -q "C:\\Users\\Hafez\\AppData\\Local\\Temp\\arduino\\sketches\\8ACE322138C4C27C313615635F30E215/partitions.csv" "C:\\Users\\Hafez\\AppData\\Local\\Temp\\arduino\\sketches\\8ACE322138C4C27C313615635F30E215/UNOR4USBBridge.ino.partitions.bin"
Error at line 8: Value 'undefined' is not valid. Known keywords: esphttpd, phy, efuse, nvs, fat, spiffs, ota, coredump, nvs_keys
```
As you pointed out, the issue may be related to the patch mentioned. I have checked the `partitions.csv` file specified in the error message, and it seems to align with the values set by the patch. Specifically, there appears to be a problem with the value in line 8, which is set to 'undefined' and is not accepted as a valid value. Here's a screenshot for reference: 

[image]

Could you please provide further guidance on resolving this issue?
Thank you.

TheHafez, 2023-09-04:
Is there a simpler method to change the VID & PID for the Arduino Uno R4? My primary aim in creating a custom firmware is to modify these values. If there’s an easier way to achieve this, I’d appreciate any guidance.

pennam, 2023-09-04:
> Hi @pennam, I've attempted to run the updated compile.sh script on a clean setup but am still encountering the same error:
>
> ```
> esptool.py v4.5.1
> Creating esp32s3 image...
> Merged 2 ELF sections [...]

This looks like a limitation/issue of the `gen_esp32part` window tool that do not support the `undefined` partition SubType. The linux script is handling this partition type without issues https://github.com/espressif/esp-idf/blob/62ee4135e033cc85eb0d7572e5b5d147bcb4349e/components/partition_table/gen_esp32part.py#L70

TheHafez, 2023-09-04:
@pennam, thanks for identifying that the issue likely stems from a limitation with the gen_esp32part tool on Windows. Since I don’t use or have experience with Linux, what can be done in my case?

pennam, 2023-09-05:
@TheHafez what if you put `0x06` instead of undefined? https://github.com/espressif/esp-idf/blob/62ee4135e033cc85eb0d7572e5b5d147bcb4349e/components/esp_partition/include/esp_partition.h#L95
otherwise i would try to use `phy` and see what happens.

TheHafez, 2023-09-05:
Hello @pennam,

Good news: Your suggestion to use '0x06' worked perfectly. I was able to compile, export, and flash the firmware successfully. Moreover, I achieved my primary goal of changing the board's VID and PID.

Bad News: there's a new wrinkle, i dont know why but the board has two set of vid and pid and its not even in the [ino file](https://github.com/arduino/uno-r4-wifi-usb-bridge/blob/0.3.0/UNOR4USBBridge/UNOR4USBBridge.ino#L143-L144) to be able to change them _(the other vid & pid are 0x2341 & 0x006D)_ the problem is the arduino doesnt accept uploads on the changed vid & pid nor the old unchanged vid & pid as this werid thing happened to me since i bought the board and tried to upload any random sketch, it just gives me no device found on COMX then it changes to another COM port with the other vid and pid that accepts uploads. it keeps swithing between the one accepts uploads and the one that doesnt accept when i try to upload a sketch and this isnt new as it was an issue before flashing the custom firmware.

TheHafez, 2023-09-05:
I saw other people report this issue as well:

https://github.com/arduino/ArduinoCore-renesas/issues/73

pennam, 2023-09-06:
@TheHafez I did not understand very well what is going on, but i suppose your issue is related to the fact you have uploaded a new firmware on the ESP32-S3 of your UNO R4 WiFi. After this you always need to unplug and re-plu your board from USB otherwhise the board will stay in ESP download mode and you wont be able to upload any sketch.

The VID-PID exposed by the board in ESP mode is: 303a:1001
The VID-PID exposed by the board with standard ESP firmware should be 2341:1002

I think the issue you have linked is a different one and only valid if you are using native USB from your sketch.

If your issue persist, please, open a new issue, since the original one has been solved. Thanks.

PR #17 (2023-09-04): Use ArduinoBLE library from arduino-libraries repository
Should fix #16