arduino/uno-r4-wifi-usb-bridge #18: Arduino UNO R4 Switches Between Two Sets of VID and PID, Causes Upload Failure

TheHafez, 2023-09-06:
My Arduino UNO R4 board initially came with two sets of Vendor ID (VID) and Product ID (PID):
- First Set: VID 2341, PID 1002
- Second Set: VID 2341, PID 006D

The board has not undergone any firmware updates prior to this issue.
### Issue:
When I attempt to upload a sketch using the first set of VID and PID (2341:1002), the upload process fails with the following error message:
```
Cannot perform port reset: TOUCH: error during reset: opening port at 1200bps: Invalid serial port
No device found on COM6
Failed uploading: uploading error: exit status 1
```
Immediately after the failure, the board automatically switches to the second set of VID and PID (2341:006D) and changes the COM port (for example, from COM6 to COM2). Upon retrying the upload, it successfully completes on this second set.

## Additional Information:
`I use Arduino IDE 2.2.1`
Recently, I flashed the firmware to change the first set of VID and PID. However, the [.ino file](https://github.com/arduino/uno-r4-wifi-usb-bridge/blob/0.3.0/UNOR4USBBridge/UNOR4USBBridge.ino#L143-L144) did not contain information for the second set (2341:006D) to change them. Consequently, the board continues to toggle between the modified and original VID and PID sets. As a result, I am still unable to upload sketches using the modified first set.
This issue appears to be common as I've seen multiple reports of similar behavior.
Please assist in resolving this problem.

alranel, 2023-09-06:
Hello @TheHafez, can you post the full code of your sketch?
Also, can you test it with the original unmodified firmware?

TheHafez, 2023-09-06:
Hello @alranel ,

Here is the full code of my Arduino sketch:
```
#include <Mouse.h>

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ; // Wait for serial port to connect
  }
  Mouse.begin();
}

void loop() {
  if (Serial.available()) {
    String command = Serial.readStringUntil('\n');
    processCommand(command);
  }
}

void processCommand(String command) {
  if (command.length() > 0) {
    if (command[0] == 'M') {
      int commaIndex = command.indexOf(',');
      if (commaIndex != -1) {
        int x = command.substring(1, commaIndex).toInt();
        int y = command.substring(commaIndex + 1).toInt();
        moveMouse(x, y);
      }
    }
    else if (command[0] == 'C') {
      clickMouse();
    }
  }
}

void moveMouse(int x, int y) {
  Mouse.move(x, y);
}

void clickMouse() {
  Mouse.click(MOUSE_LEFT);
}
```
Regarding your second question, I've already tested this issue with the original, unmodified firmware. The problem was present from the beginning.
Thank you for your attention to this matter.

alranel, 2023-09-06:
When you use a HID library such as Mouse.h, that library takes control of the USB port. In order to upload a new sketch, you need to put the board back into programming mode by pressing the reset button twice quickly. Can you try that?

TheHafez, 2023-09-06:
Hello,

You are correct; I noticed that this issue specifically occurs when I use a HID library like Mouse.h. I've tested uploading other sketches that don't include a HID library, and the board maintains the same VID and PID throughout the process, confirming your observation.

To give you a clearer understanding of the problem, I have recorded a video with the original, unmodified firmware. I'm attaching it to this message so you can see the issue in action.

I also tried your suggestion of putting the board back into programming mode by pressing the reset button twice quickly. While this did allow me to upload new sketches, the board still switched to the VID 2341 and PID 006D, which is not what I want, as I can only change the VID 2341, PID 1002 for the firmware.

Looking forward to your further guidance on this issue.
Thank you.

https://github.com/arduino/uno-r4-wifi-usb-bridge/assets/143944414/0309ec57-f255-40c3-9770-fafba98ce3a2

alranel, 2023-09-06:
Customizing VID and PID is not an intended feature of the UNO R4 so we can't provide support and you'll have to refer to the data sheet of the ESP32-S3 module yourself to understand how to hack it and achieve what you want.

closed 2023-09-06 by alranel

TheHafez, 2023-09-06:
FUCK YOU THEN