arduino/uno-r4-wifi-usb-bridge #19: BossaArduino.h: No such file or directory

JAndrassy, 2023-09-12:
```
In file included from ..\at_handler.cpp:14:
..\cmds_ota.h:6:10: fatal error: BossaArduino.h: No such file or directory
 #include <BossaArduino.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
```
is it some private library?

pennam, 2023-09-12:
@JAndrassy  Is it ? Code comes from here https://github.com/pennam/BOSSA/commits/esp-library and i thought it is public. I need to mainline the changes into the `Arduino` org

pennam, 2023-09-12:
the compile script should take care of downloading the library and checkout the correct branch

https://github.com/arduino/uno-r4-wifi-usb-bridge/blob/65101b8987733c1f9d19672707c4440e2fa855eb/compile.sh#L20

JAndrassy, 2023-09-12:
sorry and thank you. I didn't use the script and search on GitHub somehow didn't find the library.
I made it compile in IDE to fix some errors in the AT part

closed 2023-09-12 by JAndrassy