arduino/uno-r4-wifi-usb-bridge #50: SSL Connection Issue After Arduino R4 WiFi Firmware Update to 0.4.1

pdo59, 2024-03-03:
Issue Description:
After updating the Arduino R4 WiFi firmware to version 0.4.1, SSL connections no longer function properly. My sketch establishes a connection to a server on port 443 using the WiFiSSLClient, but the connection fails. Reverting the firmware version back to 0.3.0 restores functionality to the sketch.

Steps to Reproduce:
1.Update Arduino R4 WiFi firmware to version 0.4.1.
2.Execute sketch attempting to establish an SSL connection to a server on port 443 using WiFiSSLClient.
3.Observe that the connection fails to establish.

Expected Behavior:
The sketch should successfully establish an SSL connection to the server using the WiFiSSLClient library, regardless of firmware version.

Additional Information:
Reverting the firmware to version 0.3.0 allows the sketch to function correctly, indicating a potential compatibility issue or bug introduced in firmware version 0.4.1.

pennam, 2024-03-04:
Hi @pdo59,
loading the debug version of the firmware shoudl give us more information about why the connection is failing. You can fing the binaries here:
https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/download/0.4.1/unor4wifi-debug.zip
Tp replicate the issue i also need to know which server are you trying to connect to and the sketch you are using.

pdo59, 2024-03-05:
Ok, next weekend I will take the test and send you the test results. Thank you

pdo59, 2024-03-16:
Hi pennam,
which file among those contained in the .zip contains the debug firmware that I need to use?

pennam, 2024-03-18:
@pdo59  `S3-ALL.bin` Let me know if you need firther help

pennam, 2024-11-20:
@pdo59 I'm closing this, feel free to reopen whenever you have some feedback about your issue.

closed 2024-11-20 by pennam