arduino/uno-r4-wifi-usb-bridge #52: Compilation error: conflicting declaration 'USBCDC USBSerial'

Exryz, 2024-05-26:
I had problems with running the compile.sh script where I get this error:
```
bugUtils.h: No such file or directory
 #include "Arduino_DebugUtils.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.bugUtils.h: No such file or directory
 #include "Arduino_DebugUtils.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

but when I tried compiling through the Arduino IDE, it goes through the error but I end up with a new one
```
Compilation error: conflicting declaration 'USBCDC USBSerial'
```

here's the full error
[error.txt](https://github.com/arduino/uno-r4-wifi-usb-bridge/files/15447203/error.txt)

Exryz, 2024-05-26:
Update, after redoing everything and applying the patch manually to my packages folder where my platforms are installed. Im now having an entirely new problem. This error is being made through the compile.sh script. The same error above persists when trying to compile through the Arduino IDE.

```"C:\\Users\\kisek\\AppData\\Local\\Arduino15\\packages\\esp32-patched\\esp32/tools/esptool/esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB --elf-sha256-offset 0xb0 -o "C:\\Users\\kisek\\AppData\\Local\\Temp\\arduino\\sketches\\8BB521904647DA7EC7390AE5BEA2B649/UNOR4USBBridge.ino.bin" "C:\\Users\\kisek\\AppData\\Local\\Temp\\arduino\\sketches\\8BB521904647DA7EC7390AE5BEA2B649/UNOR4USBBridge.ino.elf"
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
"C:\\Users\\kisek\\AppData\\Local\\Arduino15\\packages\\esp32-patched\\esp32/tools/gen_esp32part.exe" -q "C:\\Users\\kisek\\AppData\\Local\\Temp\\arduino\\sketches\\8BB521904647DA7EC7390AE5BEA2B649/partitions.csv" "C:\\Users\\kisek\\AppData\\Local\\Temp\\arduino\\sketches\\8BB521904647DA7EC7390AE5BEA2B649/UNOR4USBBridge.ino.partitions.bin"
Error at line 3: Value 'undefined' is not valid. Known keywords: esphttpd, phy, efuse, nvs, fat, spiffs, ota, coredump, nvs_keys
```
```Used platform       Version Path
esp32-patched:esp32 2.0.9   C:\Users\kisek\AppData\Local\Arduino15\packages\esp32-patched\esp32
Error during build: exit status 2
```

per1234, 2024-05-26:
```text
Error at line 3: Value 'undefined' is not valid. Known keywords: esphttpd, phy, efuse, nvs, fat, spiffs, ota, coredump, nvs_keys
```

The error is specific to Windows. The problem is that when compiling on a Windows machine the "esp32" platform is configured to use the executable `tools/gen_esp32part.exe` instead of `tools/gen_esp32part.py` (which is used on Linux and macOS machines):

https://github.com/arduino/arduino-esp32/blob/8ef25a73ff8cc175836b0f65a6b276d8aed324bd/platform.txt#L27

```text
tools.gen_esp32part.cmd=python3 "{runtime.platform.path}/tools/gen_esp32part.py"
tools.gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"
```

The `espressif/arduino-esp32` repository maintainers did not keep `tools/gen_esp32part.exe` in sync with changes to the `tools/gen_esp32part.py` script source code so the `tools/gen_esp32part.exe` at the submodule's https://github.com/arduino/arduino-esp32/commit/8ef25a73ff8cc175836b0f65a6b276d8aed324bd was generated from the antiquated version of the script at https://github.com/arduino/arduino-esp32/commit/5502879a5b25e5fff84a7058f448be481c0a1f73, which was prior to [the addition](https://github.com/arduino/arduino-esp32/commit/00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff) to the `tools/gen_esp32part.py` script of support for the `undefined` partition type keyword:

https://github.com/arduino/arduino-esp32/blob/5502879a5b25e5fff84a7058f448be481c0a1f73/tools/gen_esp32part.py#L51-L68

```python
SUBTYPES = {
    APP_TYPE: {
        'factory': 0x00,
        'test': 0x20,
    },
    DATA_TYPE: {
        'ota': 0x00,
        'phy': 0x01,
        'nvs': 0x02,
        'coredump': 0x03,
        'nvs_keys': 0x04,
        'efuse': 0x05,
        'esphttpd': 0x80,
        'fat': 0x81,
        'spiffs': 0x82,
    },
}
```

`tools/gen_esp32part.exe` was recently rebuilt from the up-to-date script code in https://github.com/espressif/arduino-esp32/commit/86040af7540b1e4688faeceba3063d894a88a52c so the best solution would probably be to just update the submodule.

Exryz, 2024-05-26:
Update: fixed the current problem and was able to run the compile.sh script without any errors. Went to the unor4wifi.csv file and left the subtype blank instead of 'undefined'

ezra-ch, 2024-07-14:
stuck  stuck at exit status 2, also on windows. Cant compile.

Also what csv file are you referring too? i dont see any csv files.

ezra-ch, 2024-07-14:
i managed to compile after doing what per1234 suggested. Thanks!

closed 2024-11-20 by pennam