arduino/uno-r4-wifi-usb-bridge #63: Prerelease validation

andreagilardoni, 2024-11-20:
# Summary
The purpose of this issue is to collect all the results of test execution and validate the changes introduced in the code before releasing the next increment in firmware

# Changes introduced

## Firmware side
- https://github.com/arduino/uno-r4-wifi-usb-bridge/pull/53: add a command to make ota download non blocking and retrieve information related to the progress of ota
- https://github.com/arduino/uno-r4-wifi-usb-bridge/pull/54: add a command to retrieve the version of the firmware in a uint32_t variable
- https://github.com/arduino/uno-r4-wifi-usb-bridge/pull/62: improvements on preferences api, see PR for more details

## Core side
- https://github.com/arduino/ArduinoCore-renesas/pull/325: Core changes for #53
- https://github.com/arduino/ArduinoCore-renesas/pull/349: improvements in AT parser that enables an higher reliability and fix some issues occurring in the parse of the incoming commands
- https://github.com/arduino/ArduinoCore-renesas/pull/393: Core changes for #62 

# Procedure

In order to validate the version increment we are going to:
1.  apply all the aforementioned PR to the firmware at commit aef6457c0f324686177a45792d62aa7ba27df8ce
and we are going to use the core at commit https://github.com/arduino/ArduinoCore-renesas/commit/1169821e942f2618336b51e7a76be2be7c68f9ba
1. run all the examples in the following directories: 
    - https://github.com/arduino/ArduinoCore-renesas/tree/main/libraries/WiFiS3/examples
    - https://github.com/arduino/ArduinoCore-renesas/tree/main/libraries/SoftwareATSE/examples
    - https://github.com/arduino/ArduinoCore-renesas/tree/main/libraries/Preferences/examples

andreagilardoni, 2024-11-21:
Test name      | Compilation | Execution
---------------|---------    | ----
Prefs2Struct   | OK          | OK
StartCounter   | OK          | OK

andreagilardoni, 2024-11-21:
Test name      | Compilation | Execution
---------------|---------    | ----
ArduinoCloudSketch   | OK          | OK
ArduinoCloudOTA       | OK          | OK

pennam, 2024-11-25:
pre-released as 0.5.0 https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/tag/0.5.0

closed 2024-11-25 by pennam